ottowayi/pycomm3 #191: [FEATURE] -

SISAutomationIMA, 2021-11-22:
Hello,

Tags on Logix Rockwell controllers are not case sensitive.
https://literature.rockwellautomation.com/idc/groups/literature/documents/pm/1756-pm004_-en-p.pdf#page=26

Would it be possible to add a feature so that there is no case sensitivity on reading or writing tags ?

The problem is that the tags are stored in a dictionary. 
The keys of this dictionary are the name of the tags. Unfortunately dictionaries keys are case sensitive.

Maybe it would suffice if the key of this dictionary is always in low case or else implement a case insensitive dictionary.

ottowayi, 2021-11-22:
yeah this has been a feature I've wanted to add for awhile, I haven't yet because it's a pretty complicated.  I'm currently working on a couple other features, but will try and include this as well.

blauvster, 2025-01-15:
Here is the function I use to do this:
`    
def get_tag_case(driver, tag_name):
        tree = tag_name.lower().split(".")

        # check if program tag, if so combine the first 2 of tree
        if tree[0][:8] == "program:" and len(tree) >= 2:
            tree[0] += "." + tree[1]
            del tree[1]

        name = ""
        parent = driver.tags_json
        for branch in tree:
            branch_parts = branch.split("[")
            for tag, data in parent.items():
                if branch_parts[0] == tag.lower():
                    # add .
                    if len(name) > 0:
                        name += "."

                    # add actual tag name
                    name += tag

                    # add array info
                    if len(branch_parts) > 1:
                        name += "[" + branch_parts[1]

                    if 'internal_tags' in data['data_type']:
                        # tag has child objects
                        parent = data['data_type']['internal_tags']
                    elif len(tree) >= 2 and \
                            id(branch) == id(tree[-2]) and \
                            data['tag_type'] == 'atomic' and \
                            tree[-1].isnumeric():
                        # an atomic tag and bit-address
                        return name + "." + tree[-1]
                    elif id(branch) != id(tree[-1]):
                        # not the last part of the tag string so could not find
                        return False
                    break
        return name
`