ottowayi/pycomm3 #207: Microo850 UDTs

VeryCurious1, 2022-02-21:
Hi,

I am new to Github and Python, however I did come across your pycomm3 work with AB PLCs. I have downloaded Python3 and pycomm3 and have it set up and running. I have started testing and need to do a lot more reading to get up to speed, however, so far pycomm3 looks very promising and I was actually able to use it (as a newbie) without too much effort for pulling some Micro800 tag values!

So far, I have been able to successfully read various simple type tags, however, my Micro850 program uses several UDTs to make managing the data much simpler. My questions are: 
1) Does pycomm3 support reading / writing structures from the Micro800 PLC? (I have looked and tried the example but I get a message that the "Tag doesn't exist".
2) Also, does pycomm3 support uploading of all global tags from the Micro800 PLC?

Thank you for your support and for developing pycomm3! I hope I have submitted this question properly. I see that I will also have to do some more reading about GitHub.

ottowayi, 2022-05-16:
Hey, I apologize for not responding to this issue sooner, I don't know how I missed it.  I haven't specifically tested structs with Micro800's mostly because it's not supported in the free version of CCW.  I would think they'd work if they behave like other Rockwell PLCs, but these stupid controllers do just enough things differently to be a pain.  
 I know some of the system tags that start with `__` are not included in the upload.  When the driver uploads the tag list, we ask the PLC to tell us all the tags it has and for some reason those and others (like arrays not starting at 1) are excluded from that list by the PLC.  Unfortunately that means the driver doesn't support them then.

Thanks for submitting this issue, you did everything correct.. but, I failed to respond in a timely manner, sorry!

ottowayi, 2022-05-18:
@Ajility97 

Hardware is an issue for me too now that I'm no longer with an integrator my test devices are limited to what I purchase.  I currently have a really old compact logix and a micro 1100.  Unfortunately I don't have a suggestion besides searching ebay.  

For Micro800s, I use the simulator to test against which has been fine for the most part.  The only downside is that I only have the free version of CCW, which doesn't allow creating of UDTs.  I would like to add struct support or fix whatever bug there currently is, but I need something to test against.  I've been thinking about how to go about it and I'm wondering if someone has a sample program export then I may be able to load it into my simulator and test via that.  If you're able to make one for me (or @VeryCurious1 if you're still watching this issue) that would help a lot.  I'm also going to reach out to some of my contacts and see if they would be able to as well.

Ajility97, 2022-05-18:
@ottowayi 

Is [ian@ottoway.dev](mailto:ian@ottoway.dev) a good email for you? I'd like to help.

ottowayi, 2022-05-18:
Yep!

VeryCurious1, 2022-05-18:
Hi Ian,

I am out of the office this week with very limited email access. I see that there are a couple of emails that have been sent. I will review them and get back to you by the middle of next week – at least with some more information or sample files.

Just off the top of my head, the __ are System and I/O tags. They can be read / and write tags can be written (but they are not exported from CCW). CCW Developer edition is required to create User Defined Tags UDTs - structures. Using Ethernet/IP services 0x4C and 0x4D services (and 0x52 and 0x52 fragmented) one can read / write all the PLC tags including arrays,  which does not require the CCW developer edition.

Micro800s using symbolic addressing (similar to Logix controllers) and are slowly replacing the MicroLogix controllers. Micro 8xxs  only support a subset of ControlLogix / CompactLogix services (Symbolic Segment Services using the ANSI Extended Symbolic Segment in the service path.

Have you implemented any of these services with Logix PLCs?

VeryCurious1, 2022-05-24:
Hi Ian,

I may be able to help out. Let me know if you are still interested?

I have a couple of questions to better understand how I can best help:

  1.  Is Pycomm3  currently able to read symbolic tags from a Compactlogix PLC? Are you using CIP services 0x4C and 0x4D?
  2.  It sounds like you don’t have a Micro800 PLC but you have the free version of CCW. Please confirm the version.
  3.  If you don’t have a recent version of CCW, you can download a  free version of CCW from the Rockwell website. (I am using CCW V12 Developer). This version and perhaps V11 have a Micro850 Simulator built in which you can use for testing purposes – no need for actual hardware. (It runs locally on your computer (using the local IP address) for up to one day, then you would need to restart it. Have you ever used CCW with the built in Micro850 Simulator?
  4.  At this point you could create some tags in CCW and download them to the simulator for testing the Micro850 Simulator with Pycomm3. You should be able to test all “simple” tag types and arrays. This would be the first step. (If you need some help, then let me know).
  5.  Once the above is working, I could generate some structured tags (UDTs – User Defined Tags) for you. (The Python code changes would be quite simple for UDTs as it is very similar to reading regular tags. You could load the file and test these UDTs  (read and write) in the free version of CCW, however you will not be able to change the structure of the UDTs.

Please let me know how you would like to proceed.

ottowayi, 2022-05-24:
@VeryCurious1 

1 - Yeah symbolic reads are supported, but it will use the instance IDs where supported.  For example on a Compact base tags use instance ids but program-scoped ones are all symbolic, but you can also force it to use symbolic if needed.  The read/write and fragmented services are all implemented, along with multi-service requests (but that is disabled on Micro800s since they don't support it). 
2 - Correct, I don't have a physical device but I use the simulator for testing and that's working well.
3 - yep, I'm using the free version of CCW v20, I think the Micro800 firmware version is 12. 
4 - yep, that's working.
5 - this is where I'm stuck, the free version doesn't let me create UDTs, but if you're willing to send me an example program with some already defined I believe that's all I'll need.  Particularly,  UDTs with boolean arrays and attributes with different data type sizes since sometimes it leads to padding bytes being added if the data isn't word-aligned.   I have full structure (UDT, AOI, and predefined) reading and writing working for Control/CompactLogix, but I'm not sure what the issue is on the Micro800.   

So yeah, if you're willing to send me a sample program with some example UDTs that would be great and help me out a lot.   Thanks!

VeryCurious1, 2022-05-24:
Hi Ian,

OK, that sounds good. I am still on V12 (V13 didn’t have many improvements), and V20 had several issues and was recently pulled from the website but is not back.

  1.  So it sounds like you are you able to read / write simple:  a) tags and b) arrays. I have included some samples.

(If you can, that is great. I think that the rest should be fairly straight forward. I think it will just involve some iteration of the “.” dot separators in the Path Segment 0x91).

  1.  I have included some simple arrays i.e. DIntArray, RealArray, and StrArray. (You can create these arrays in CCW as it does not require the Developer edition).

  1.  I have also included 4 UDTs (which do require CCW Developer edition to make any changes) to test various functionality: i.e.  TestSimple (simple structure), TestArrays (structure of arrays),   WO_Single (a more complex structure), WOs (an array of structures). Let me know if you want me to change any of the structures. (Note that to include an array in a UDT, an array needs to be created as a Data Type. For instance, you can’t insert DIntArray into a UDT. You must create a UDT Array, i.e Dint10Array, and then insert it into a UDT such as TestArrays. Not critical, just an FYI in case you are wondering).

  1.  You should be able to browse the structures in the global variables. I have give some initial values to some of the structures.  To access specific UDTs use for example:  WOs[3].SKU.Rate, WO_Single.SKU.Name, Test.B32[5], etc.. I have also exported the global variables into a CSV file in case you need this.

Hope this helps. Let me know if you need anything else or have any questions. My part was the easy part. Now your work begins. 😊

Ajility97, 2022-05-24:
@VeryCurious1 It doesn't look like the file attached - I'm not sure you can send attachments through email

VeryCurious1, 2022-05-24:
Ian, how should I send this?

Did you read the text in the email?

Two files were attached: Excel and the CCW project archive.

Ajility97, 2022-05-24:
@VeryCurious1 I'm not Ian, but if you're OK with sharing the program publicly, I'd suggest attaching to this issue via GitHub's website (https://github.com/ottowayi/pycomm3/issues/207) 

If not, you can email Ian directly ([ian@ottoway.dev](mailto:ian@ottoway.dev))

VeryCurious1, 2022-05-24:
Thanks Agility97. Github doesn't allow me to to post this type of file. I will send it to Ian assuming that he will be able to open it.

ottowayi, 2022-05-31:
@VeryCurious1 thanks, it worked exactly as we expected and I was to do some initial testing.  It appears the Micro800s don't include the structs in the tag list.  After connection, the driver uploads the tag list and parsers their definitions to get things like attributes, their data types, and other information about how the data is packed.  These definitions allow the structure reading/writing to work, so if they tags are excluded from the list the driver thinks they don't exist.  I've noticed they also excluded arrays that don't start at 0.  

There is a workaround, but it's not easy.  You can manually define the tags, but it's not documented as to what fields are required and when.   Unfortunately, I tried manually defining the struct, but was met with an _unknown/unsupported_ error.   This most likely means the full structure can't be accessed at once and will instead require doing each attribute individually. 

Here is what I did to test:
```python
from pycomm3 import LogixDriver, configure_default_logger, LOG_VERBOSE, Struct, STRING, DINT, REAL
from rich import print
configure_default_logger(level=LOG_VERBOSE)

SKU_Struct = Struct(DINT('No'), STRING('Name'), REAL('Rate'), DINT('CartonQty'), DINT('CaseQty'))
OEE_Struct = Struct(REAL('Avail'), REAL('Perf'), REAL('Qual'), REAL('OEE'))
WO_Struct = Struct(
    STRING('StartDate'),
    SKU_Struct('SKU'),
    DINT('TargetQty'),
    DINT('ActualQTY'),
    OEE_Struct('Metrics'),
)

with LogixDriver('172.19.207.136') as plc:
    # print(plc.tags)
    plc.tags['WO_Single'] = {
        'data_type': {
            'name': 'WO_Struct',
            'data_type_name': 'WO_Struct',
            'type_class': WO_Struct,
            'template': {'structure_size': 100},  # not accurate, needs some value b/c it's used in estimating response size 
            'internal_tags':
                {
                'TargetQty': {
                    'tag_type': 'atomic',
                    'data_type': 'DINT',
                    'data_type_name': 'DINT',
                    'type_class': DINT,
                }
            }
        },
        'tag_type': 'struct',
        'elements': 1,
    }
    responses = plc.read('WO_Single', 'WO_Single.TargetQty')

print(responses)

```

Only the read of `WO_Single.TargetQty` works, so to add other attributes they'll also need to be defined under `internal_tags`.  Right now it's kinda messy, like the 3 different type fields, so it can be a pain to manually create the definitions.  I plan on making this easier in v2, by consolidating some of the fields, using dataclasses instead of raw dicts, etc.  But I don't have a timeline on when it will be ready, I've started working on it but I haven't had much time to work on it lately.

closed 2023-03-26 by ottowayi