ottowayi/pycomm3 #208: Is there a method to get a complete tag list with a breakdown of UDT's?

Destination2Unknown, 2022-02-25:
Is there a simple way to get a complete tag list with a breakdown (in a simple format to send to a csv file) of the structure? 
Something like:

```
SimpleTag - DINT
                              
ComplexTag - Cfg - LoLimit - Real
                 - HiLimit - Real
           - Sts - Start - DINT
                 - End - DINT
                 - Complete - Bool
```
                                   

                              
Thanks

ottowayi, 2022-02-25:
Sort of, there is the `LogixDriver.tags` property which is a dictionary of all the tag names to definitions, but there is not a builtin way to get just the types.  There's also the `LogixDriver.tags_json` which filters some of the fields that aren't JSON-supported.  If you want just the types you'll have to filter it yourself, but you can base it off the code for `tags_json`.  [This page](https://docs.pycomm3.dev/en/latest/usage/logixdriver.html) in the docs lists all of the fields available.

Destination2Unknown, 2022-02-26:
Was thinking that, nothing is straightforward when it comes to udt's. Cheers

closed 2022-02-26 by Destination2Unknown