ottowayi/pycomm3 #210: Moving from a routed CIP Message to a unrouted CIP Message

JL00001, 2022-03-07:
I am using pycomm3 to write parameters for PowerFlex 525 Drives. I based how I am doing this off of your generic messaging examples here:
https://docs.pycomm3.dev/en/latest/examples/generic_messaging_examples.html

My question is this:
 Do I need to use a CIP routing string "drive_path = '10.10.10.100/bp/1/enet/192.168.1.55'" or can I use just use the drive's IP address?
I have played around the route_path=True argument but with no luck. I keep getting 
```Generic message 'forward_open' failed: Service not supported```
and 
```forward_open failed - Service not supported```

This is the code I am using:
```
with pycomm3.CIPDriver("192.168.100.33") as pf525:
        result = pf525.generic_message(
                service=pycomm3.Services.get_attribute_single,
                class_code=b'\x93',
                instance=29,
                attribute=b'\x09',
                data_type='INT',
                connected=True,
                unconnected_send=True,
                route_path=True,
            )
    print(result.value)
```

JL00001, 2022-03-07:
```
2022-03-07 16:35:00|pycomm3.cip_driver.CIPDriver| ERROR   | Generic message 'forward_open' failed: Service not supported
2022-03-07 16:35:00|pycomm3.cip_driver.CIPDriver| ERROR   | forward_open failed - Service not supported
5001
```

ottowayi, 2022-03-07:
What's causing the issue is `connected=True`, you'll want that to be `False` usually.   A _connected_ message uses a CIP connection that is created by the _forward open_ service, and and _unconnected_ message uses UCMM to forward messages to the target.  Connected messaging has some performance benefits, but for single requests like this you'll not see any of them.. and obviously can't use them if they're not supported.  

You'll need the full CIP route to the drive as well.  You can put the full path in the for the driver.  What happens is the route is split into the IP address and the rest of the CIP path.  So for you example above, we'd open a connection to `10.10.10.100` and then store `bp/1/enet/192.168.1.55` as the CIP path to the target.  So when you send a request, we send it to `10.10.10.100` and then tell it's UCMM to forward it on until it get's to `192.168.1.55`.   This is what `route_path=True` does, it puts the remaining CIP path into the message.  You can also open a driver with just the IP like `CIPDriver('10.10.10.100')` and then set `route_path='bp/1/enet/192.168.1.55` to get the same effect.  That method works well if you have a bunch of drives you want to do at once, you only end up needing to create one driver.   This is also when `unconnected_send` comes in, since you have a route and are using an unconnected message, we need to use a `Unconnected Send` service to tell the UCMM to forward on our message.  So you'll want that to be `True`, or else our message stops at whatever device is `10.10.10.100`.  I didn't realize the uncconected send was so simple when I originally wrote this, so whenever I get the next version ready it'll be removed and become automatic. 

TL;DR  try this:

```python
with pycomm3.CIPDriver("10.10.10.100/bp/1/enet/192.168.1.55") as pf525:
        result = pf525.generic_message(
                service=pycomm3.Services.get_attribute_single,
                class_code=b'\x93',
                instance=29,
                attribute=b'\x09',
                data_type=pycomm3.INT,
                connected=FALSE,
                unconnected_send=True,
                route_path=True,
            )
    print(result.value)

```

JL00001, 2022-03-07:
So what I am seeing is this; even with the two errors ```Generic message 'forward_open'``` and ```forward_open failed``` I am getting data from my reads and my writes are going through. I haven't tested this to my fullest satisfaction yet but, other than the error messages; it seems to work without the CIP path, just the drive IP. How is that possible?

I should mention that I am on a flat network; No routers or backplanes between me and the drive.

Also is your ```connected=FALSE``` suppose to be ```connected=False```? not all in caps?

JL00001, 2022-03-08:
That would explain this error:
```2022-03-08 10:38:20|pycomm3.cip_driver.CIPDriver| ERROR   | Generic message 'generic' failed: Too much data```

Log:
```
2022-03-08 10:38:20|pycomm3.cip_driver.CIPDriver| DEBUG   | Opening connection to 192.168.100.33
2022-03-08 10:38:20|pycomm3.cip_driver.CIPDriver| DEBUG   | Sent: RegisterSessionRequestPacket(message=[b'\x01\x00', b'\x00\x00'])
2022-03-08 10:38:20|pycomm3.cip_driver.CIPDriver| DEBUG   | Received: RegisterSessionResponsePacket(session=41, error=None)
2022-03-08 10:38:20|pycomm3.cip_driver.CIPDriver| INFO    | Session=41 has been registered.
2022-03-08 10:38:20|pycomm3.cip_driver.CIPDriver| INFO    | Sending generic message: generic
2022-03-08 10:38:20|pycomm3.cip_driver.CIPDriver| DEBUG   | Sent: GenericUnconnectedRequestPacket(message=[b'\x10', b'\x03 \x93$>0\t', b'1\x00', b'\x00\x00'])
2022-03-08 10:38:20|pycomm3.cip_driver.CIPDriver| DEBUG   | Received: GenericUnconnectedResponsePacket(service=b'\x10', command=b'o\x00', error='Too much data')
2022-03-08 10:38:20|pycomm3.cip_driver.CIPDriver| ERROR   | Generic message 'generic' failed: Too much data
2022-03-08 10:38:20|pycomm3.cip_driver.CIPDriver| DEBUG   | Sent: UnRegisterSessionRequestPacket(message=[])
2022-03-08 10:38:20|pycomm3.cip_driver.CIPDriver| DEBUG   | Received: UnRegisterSessionResponsePacket()
2022-03-08 10:38:20|pycomm3.cip_driver.CIPDriver| INFO    | Session Unregistered
2022-03-08 10:38:20|pycomm3.cip_driver.CIPDriver| DEBUG   | Opening connection to 192.168.100.33
2022-03-08 10:38:20|pycomm3.cip_driver.CIPDriver| DEBUG   | Sent: RegisterSessionRequestPacket(message=[b'\x01\x00', b'\x00\x00'])
2022-03-08 10:38:20|pycomm3.cip_driver.CIPDriver| DEBUG   | Received: RegisterSessionResponsePacket(session=42, error=None)
2022-03-08 10:38:20|pycomm3.cip_driver.CIPDriver| INFO    | Session=42 has been registered.
2022-03-08 10:38:20|pycomm3.cip_driver.CIPDriver| INFO    | Sending generic message: generic
2022-03-08 10:38:20|pycomm3.cip_driver.CIPDriver| DEBUG   | Sent: GenericUnconnectedRequestPacket(message=[b'\x0e', b'\x03 \x93$>0\t', b'', b'\x00\x00'])
2022-03-08 10:38:20|pycomm3.cip_driver.CIPDriver| DEBUG   | Received: GenericUnconnectedResponsePacket(service=b'\x0e', command=b'o\x00', error=None)
2022-03-08 10:38:20|pycomm3.cip_driver.CIPDriver| INFO    | Generic message 'generic' completed
2022-03-08 10:38:20|pycomm3.cip_driver.CIPDriver| DEBUG   | Sent: UnRegisterSessionRequestPacket(message=[])
2022-03-08 10:38:20|pycomm3.cip_driver.CIPDriver| DEBUG   | Received: UnRegisterSessionResponsePacket()
2022-03-08 10:38:20|pycomm3.cip_driver.CIPDriver| INFO    | Session Unregistered
```

which was created by this code:

```
def __write_pf525_data(Class_Code, instance, request_data, Attribute):
            with pycomm3.CIPDriver("192.168.100.33") as drive:
                          result = drive.generic_message(
                              service=pycomm3.Services.set_attribute_single,
                              class_code=Class_Code,
                              instance=instance,
                              attribute=Attribute,
                              connected=False,
                              request_data=pycomm3.INT.encode(request_data),
                              )
```

closed 2022-03-08 by JL00001

ottowayi, 2022-03-08:
Ohh yeah, sorry that is my fault, I forgot to mention that detail about `route_path`.   You're not the first person to have the same issue and I totally forgot about it.  Right now I'm not very happy with the `generic_message` implementation, it requires the user to know all the details of CIP messaging and set the correct parameters.  I started working on a pretty significant rewrite that includes a refactor of this method to make it a bit simpler and easier to use by making some of those parameters more automatic. 

I'm glad you figured it out, but I need to get this documented and the refactor done so others don't run into the same issues.

JL00001, 2022-03-08:
Good sir, I am the LAST person to complain about documentation as my codebase doesn't have any. I too am tormented by imperfect code and constant need to refactor and change what I have already written.

I wish you luck on your rewrite, and if you ever want a laugh, check out my codebase and see how yours could be worse