ottowayi/pycomm3 #224: Software revision support

dominiksr, 2022-07-28:
I have a question about CompactLogix series controller support. I am currently using the pycomm3 library to use CIPDriver to connect to and retrieve variables from the driver. CompactLogix has the v32 software revision and some problems are occurring that are forcing me to upgrade to v33. 

Is there any limitation on which revisions this library works on? Or any known problems? Upgrading to v33 will not affect the existing correct operation of my application that uses the pycomm3 library?

Thanks for the effort you put into this tool :D

ottowayi, 2022-07-30:
No, V33 should not cause a problem.  Most features of this library are common to all versions and will most likely not break anytime soon.  I would recommend you use the `LogixDriver` instead tho, it is designed to connect to those controllers.  `CIPDriver` is very basic and really only used for messaging non-PLC devices.  Also, if you want to do anything with the tags in the PLC you'll need the `LogixDriver`.

dominiksr, 2022-07-31:
Awesome, thank you for the reply

closed 2022-07-31 by dominiksr