ottowayi/pycomm3 #230: trying to write value in String register

QuakeString, 2022-08-21:
hi, I'm tyring to write string in ST22:3
but get this error 
"too many values to unpack (expected 2)"
I used this code 

with SLCDriver('192.168.1.10') as plc:
plc.write('ST21:3', 'this is simple text')

please help me out.

ottowayi, 2022-08-22:
Can you show the full stack trace or enable [logging](https://docs.pycomm3.dev/en/latest/getting_started.html#logging)?  The error that triggers that error will tell you what is wrong.

QuakeString, 2022-08-28:
I not yet got a chance to check that, I will do it tomorrow. 
but I found another thing. when I'm reading string from ST22:1 and writing it back to ST22:2 then it shows that I wrote 82 charactor, actually in ST22:1 there is only 7 charactor,
also when I read ST22:1 and store it into my sqlite db, there also storing as a BLOB.
[image]
[image]

QuakeString, 2022-08-29:
Here is the output of

> Can you show the full stack trace or enable [logging](https://docs.pycomm3.dev/en/latest/getting_started.html#logging)? The error that triggers that error will tell you what is wrong.

Here is the output:
2022-08-29 09:01:26,700 [INFO] pycomm3.cip_driver.CIPDriver._register_session(): Session=3394028776 has been registered.
2022-08-29 09:01:26,701 [INFO] pycomm3.cip_driver.wrapped(): Attempting an Extended Forward Open...
2022-08-29 09:01:26,701 [INFO] pycomm3.cip_driver.CIPDriver.generic_message(): Sending generic message: forward_open
2022-08-29 09:01:26,770 [ERROR] pycomm3.cip_driver.CIPDriver.generic_message(): Generic message 'forward_open' failed: Service not supported
2022-08-29 09:01:26,771 [ERROR] pycomm3.cip_driver.CIPDriver._forward_open(): forward_open failed - Service not supported

ottowayi, 2022-09-01:
Hmm, it looks like I didn't put in an useable logging for the SLC driver.  This driver doesn't get much attention so it's probably a bug.  I will try to investigate and figure out what is going one

mfastag, 2024-03-07:
Hi, I assume this read issue was resolved for the SLC driver (in "_slc_string_swap" I had to append a zero byte if data length is odd)....  but when I write a string, although I don't an error the value does not take. any thoughts? was the SLC driver ever successful?