ottowayi/pycomm3 #231: [BUG] - very high communication time in SLCDriver PLC Micrologix 1400 Series B

QuakeString, 2022-08-30:
PLC Micrologix 1400 Series B
Model: Bul. 1766
my code is 
with SLCDriver('192.168.1.10') as plc:
      cpl = (plc.read('B3:150/5', 'F8:5', 'F8:6', 'F8:7',
                            'F18:0', 'F18:1', 'F18:2', 'N7:150',
                            'N7:161', 'N21:16', 'B3:160/11', 'B3:160/10',
                            'B3:24/2', 'B3:160/12', 'B3:52/1', 'N21:10',
                            'N21:11', 'N21:12', 'N21:13', 'N21:14',
                            'N21:15', 'N21:23', 'L23:0', 'N21:21',
                            'N21:22'
                            ))
its taking 1.5 to 1.9 sec to cycle.
is it normal ?
here is log file
[pycomm3.log](https://github.com/ottowayi/pycomm3/files/9452788/pycomm3.log)

QuakeString, 2022-08-31:
I see a pattern that it trying to "Attempting an Extended Forward Open.." but its failing and then it trying to open as "attempting standard Forward Open"
so can I buypass the Extended forward open ? and do it as standard mode, as it is possible in LogixDriver

ottowayi, 2022-09-01:
Yeah, the both SLC and Logix drivers are subclasses of `CIPDriver`, so you should be able to do `plc._cfg['extended forward open'] = False` to disable it.  It should be disabled the SLC driver since it will never work, so that is an oversight.

ottowayi, 2022-09-01:
Also, these devices do not support multi-requests, meaning each address you do in the read method will be it's own request.  You can reduce the number of requests by reading multiple elements at a time.  Like `N21:10{10}` should return a list of 10 elements from `N21:10` to `N21:19`.

closed 2023-03-26 by ottowayi