ottowayi/pycomm3 #233: Change network port

NothinRandom, 2022-09-14:
Is there a way to change the default network port from 44818 to something else instead?  Useful if going through a proxy.  Was hoping to do something like:  `with LogixDriver(path='11.22.33.44', port=1234) as plc:`.  Thanks

ottowayi, 2022-09-15:
unfortunately it's not possible to change the port while using a context manager, but you can do:
```py

plc = LogixDriver('11.22.33.44')
plc._cfg['port'] = 1234
plc.open()
...
plc.close()
```

but, I'm going to leave this issue open because it would be nice to be able to change it without accessing a private variable.  I'll probably integrate it into the path, so you can do `LogixDriver('11.22.33.44:1234')` or something like that.

ottowayi, 2022-09-16:
actually, now this is supported in 1.2.10

closed 2022-09-16 by ottowayi

NothinRandom, 2022-09-16:
Thanks for the quick update and implementation!  Fantastic product!

ottowayi, 2022-09-16:
No problem, simple features like this I like to get done right away so I don't forget about them lol

NothinRandom, 2023-11-03:
@ottowayi,

Again, thanks for this update!

Quick question on combining this with different slot.  Would the syntax be `IP/SLOT:PORT`?  For example, `1.2.3.4/1:12345`

ottowayi, 2023-11-03:
@NothinRandom it would be `IP:PORT/ROUTE/...`
[image]