ottowayi/pycomm3 #245: Permission Denied when trying to read UDTs

johncharris, 2022-10-06:
I am able to connect to a Control Logix 1756-L81E and get the tag names and read single values, but when I try to read a UDT I get `AT_Flt1, None, None, Permission denied`.

The dumped tag info is:
"AT_Flt1": {"tag_name": "AT_Flt1", "dim": 0, "alias": False, "instance_id": 97, "symbol_address": 1879582996, "symbol_object_address": 1948063388, "software_control": 67174494, "external_access": "Read/Write", "dimensions": [0, 0, 0]

I assume this is a setting on the PLC, but I'm not sure what to look for. I'm not sure what to look at beyond the external access. Please let me know what I need to change.

Thank you

ottowayi, 2022-10-06:
Check the external access for each attribute, if there are any that are None, then trying to read the whole struct at once will result in the permission error.  In that case, you have to read each individually unless you have control over the definition and can change it.

johncharris, 2022-10-06:
I spoke with the PLC developer and it sounds like this is indeed the case. I'll just have to pull the attributes separately. Thank you!

closed 2022-10-06 by johncharris