ottowayi/pycomm3 #253: [BUG] - CIP Routing Path through newer compactlogix with 2 ethernet adapters

RobovicChristo, 2022-12-10:
**Description**
I'm trying to read a powerflex 525 parameter passing through my 5069-L320ERS2 PLC. I'm connected on Ethernet Port A1 of PLC and drive is connected on port A2. I'm getting error 'Destination unknown, class unsupported, instance undefined or structure element undefined (see extended status) - Extended status out of memory  (05, 00)'

**Target PLC**
Model:  5069-L320ERS2    
     IPs --> (A1 : 192.168.0.40)     (A2 : 192.168.1.1)

Firmware Revision: 34.011
Other Devices in CIP Path:  
Powerflex 525 v7.001      
IP : 192.168.1.30

**Code Sample**

```python
         import pycomm3

         with pycomm3.CIPDriver('192.168.0.40/4/192.168.1.30') as drive:
            Result = drive.generic_message(
                service=pycomm3.Services.get_attribute_single,
                class_code=b'\x93',
                instance=41,
                attribute=b'\t',
                data_type=pycomm3.INT,
                connected=False,
                route_path=True,
            )
```

**Additional context**
   - When connecting on port A2 of PLC with my PC and putting "192.168.1.30" as path the communication works
   - I can read the parameter via explicit messaging from another AB PLC with this path "3, 192.168.0.40, 4, 192.168.1.30"
   - I tried both "route_path=True" and "route_path=False". I'm not sure what this does. 
   - The older PLCs with only 1 Ethernet Adapter would use 2 as port but newer PLC with two adapters uses 3 for A1 and 4 for A2. I'm guessing that's what I have to enter in the routing path instead of the "enet" (which is being translated to 2 in the code).

ottowayi, 2022-12-10:
Would you be able to share a [log](https://docs.pycomm3.dev/en/latest/getting_started.html#logging)?  What you have looks right, you are correct in the `enet` alias is for port 2, I don't have aliases for ports 3/4 (yet). The `route_path` you have is correct, it basically means if you are using an unconnected message, use the route of the driver.  Without it an unconnected message would go only the first device in the path.  This is confusing, and I eventually will have a much more user friendly method in the v2 version I'm working on.  `generic_message` was originally only a private internal method and meant for finer control of the messages, but with that requires a lot more understanding of how exactly CIP works than it should.

Could you also try setting `connected=False`?  I seem to remember one time I had a drive that would only work if used a _connected_ connection.  If that doesn't work, the log should give me a better idea as to what's wrong.

RobovicChristo, 2022-12-10:
This did work ! I tried so many different combinaison with the port number, the route_path, the connected and the unconnected_send, I thought I tried them all but I missed that one. 

Thanks for your help !

closed 2022-12-10 by RobovicChristo