ottowayi/pycomm3 #265: Ability to scan for valid CIP objects

jamesbraza, 2023-02-14:
I am dealing with an [Applied Motion MDXL62GN3IPB000](https://www.applied-motion.com/products/integrated-servo-motors/mdxl62gn3?pn=MDXL62GN3IPB000) servo motor ([communications manual](https://appliedmotion.s3.amazonaws.com/Host-Command-Reference_920-0002W_0.pdf)).  I am able to use `CIPDriver._list_identity` to query the identity object.

I am using `CIPDriver.generic_message` to try and read/write with the motor's parameters, but for every combination of class code and service, I am getting errors:

```python
>>> driver = CIPDriver(MOTOR_IP_ADDRESS)
>>> driver.__enter__()
>>> driver.generic_message(
...     service=Services.get_attribute_single,
...     class_code=0x04,
...     instance=0x00,
... )
Tag(tag='generic', value=b'', type=None, error='Destination unknown, class unsupported, instance undefined or structure element undefined (see extended status) - Extended status out of memory  (05, 00)')

>>> driver.generic_message(
...     service=Services.get_attribute_single,
...     class_code=0x65,
...     instance=0x00,
... )
Tag(tag='generic', value=b'', type=None, error='Object does not exist')
```

Is there a functionality in `pycomm3` to "scan" for usable CIP objects given a particular class code?

ottowayi, 2023-02-18:
I don't believe there is a way to get all of the supported objects and there isn't a method built in to do it, so you'd probably have to set up a loop and just try every class code.  But, I don't think you need to do that.  You're using the `get_attribute_single` service, but not specifying an attribute to get.  Try using `attribute=3` in the request, that's the instance could for the class.  Since you have `instance` set to 0, that means you're accessing the class-level attributes.  Other attributes you could try are 1 for the object revision or 2 for the max instance, none of these attributes are required but I think almost all objects support them.  If you confirm the class exists, you could then try reading attributes from the specific instances.

jamesbraza, 2023-02-21:
Okay thanks!

> you'd probably have to set up a loop and just try every class code

In my code snippet above I had no `attribute` specified, but actually in local testing, I already did the suggested sweep (sweeping over hundreds of possible combinations using nested `for` loops).

I was curious if there's some `pycomm3` built-in support for this, I am okay making DIY efforts on this front.

Thanks again for responding!  Closing this out.

closed 2023-02-21 by jamesbraza