ottowayi/pycomm3 #267: Using write() to overwrite a physical input (LogixDriver)

PoisonS1K, 2023-02-23:
Hello,
A project I am trying to do is control several pushbuttons remotely. From what I can see, trying to write to a button tag with a physical location (ex: Local:2:I.Data.0) does not work as I can trend it and see no change. However, if I write to a tag that is OTE from this button tag, I can see spikes in the trend where it does read, however this is not reliable and has to be looped many times to account for variation in PLC scan time. Also, the logic that is all derived from that OTE tag does not always reliably evaluate. Is there a better way to go about this, other than remapping and adding virtual tags that are toggleable?
Thank you.

nbrown-PCC, 2023-02-23:
your ladder logic likely has your example tag (local:2:I.Data.0) driving an output of another tag (lets call it button pushed). say you want to change the value of button pushed, without referencing the actual push button (or the input from it)...

it'd be best to have your code write to a tag that disables that rung, and enables a "remote button" wrung... that's how i would do it. then you COULD have another rung that if the push button is pressed on location, it disables your remote rung and enables your local rung. if you have an alarming system or an HMI, you could then display which has control.

ottowayi, 2023-02-25:
Yeah, I think you're seeing a timing issue with the PLC overwriting the tag before you can see your value.  I believe I/O tags are updated at whatever rate the module is configured for independent of the scan time of the logic.  If you're trying to override a physical button input you'll probably need a separate tag.  In the past I would do something like this:

```
---+------] [-------( U )------+-------(  )------
   |     hmi_pb    hmi_pb      |      pb_cmd
---+-----] [------[ONS]--------+
       pb_input     

-----] [--------{logic driven by button}
    pb_cmd
```

`hmi_pb` is a _push button_ tag that's written to from an HMI or elsewhere like pycomm3.  It's immediately unlatched and drives the `pb_cmd` coil.  Then `pb_input` is a physical pushbutton I/O point that also drivers the `pb_cmd` coil with a oneshot.  This way `pb_cmd` is only ever high for a single scan each time the button is pressed.  You would use `pb_cmd` to drive any logic for the button and allows you to write to `hmi_pb` without worrying about setting it back low or it staying high for multiple scans.  

I'm not sure if this helpful (or how accurate it is, been a couple years since I've written any ladder).

closed 2023-03-26 by ottowayi