ottowayi/pycomm3 #277: Sending connected message via forward open

tlf30, 2023-05-14:
Hello,
First thank you for this amazing library, I have been using it for the last few years on small projects and it has made my life so much simpler.

I have hit a situation where I am trying to access the CIP PortObject class (0xF4) across a backplane. I have a test setup at home right now where I have the following:
13 Slot Chassis:
Slot 0: 1756-L55
Slot 5: 1756-ENET
Slot 6: 1756-EWEB - 192.168.5.11

I would like to send a Get All Attributes (0x01) for the PortObject (0x01) with Instance (0x00) to get the class attributes.
I can send the message as a connected message from my computer to the EWEB directly over the network, and it responds with no error and the class information. For reference, if using an unconnected message, it responds with error 0x01 ext 0x0205

If I send an unconnected message with the request to `192.168.5.11,1,0` or any module in the chassis including the EWEB itself, I get error 0x05.
If I use a connected message with the same request across the backplane (which uses the forward open) I get the same error 0x05.

```
data.driver = CIPDriver('192.168.5.11/1/0')
    try:
        data.driver.open()
    except CommError as e:
        print('Failed to connect to device')
    with data.driver as plc:
        result = plc.generic_message(service=0x01, class_code=0xF4, instance=0x00,
                                     attribute=0x00,
                                     request_data=b'',
                                     connected=True, unconnected_send=False, route_path=False
                                     )
        print(result.error)
        print(result.value)
```

I'm not exactly sure the internals of the forward open, I spend all day yesterday researching how it works.
What I think I need is a way to send a connected message from the EWEB to other modules on the backplane. It seems like the forward open should accomplish that, but it is not. I have tested with some of my other test tools, and have concluded that pycomm3 behaves the same way that they do, and I get the same results with them as well. 

My question is, is there a way to do what I am trying to do? I would have thought sending a connected message through a routed path would not have been difficult, but it seems like I cannot find a way to do it. Although, I do not fully understand the difference between the UCMM, connected and unconnected messages. 

Any help would be much appreciated,
Trevor Flynn

ottowayi, 2023-05-17:
There's a couple things could going on here.  First is that error means the object/instance/attribute doesn't exist on the device you're talking to.  This isn't too surprising with these older modules and the Port object specifically.  I had a issue with some code at work that relied on the port object, but found on older firmwares for ENBT/ENET cards it just didn't work.  Also, some don't support get_attributes_all, so had to fall back to doing individual get_attribute_single on each attribute, but you would see a different _service not supported_ error code (so be aware of that if you come across it). 

The second could be the routing, so I'll try and explain how it works.  If `connected=True`, `generic_message` uses the forward-open connection for the driver, in the example to your L55.  All connected messages will be routed to slot 0 since that's what the driver was setup as.  But, with `connected=False` the request is routed to whatever `route_path` points to, when that is `True` it reuses the route from the driver.  Problem is that if you have a route beyond what you're connected to (in this case we're connected to `192.168.5.11` and a route to `backplane/0`) the remaining route needs to be wrapped in an unconnected send or the target's UCMM won't be able to forward the request on.  In this case, you'll need to set `unconnected_send=True`.

Here's a small example to demonstrate:
```py
from pycomm3 import CIPDriver, ClassCode, Services, SHORT_STRING

def test(connected, route_path, unconnected_send):
    with CIPDriver('10.1.128.45') as driver:
        print(f'Running with: {connected=}, {route_path=}, {unconnected_send=}...')
        resp = driver.generic_message(
            service=Services.get_attribute_single,
            class_code=ClassCode.identity_object,
            instance=1,
            attribute=7,
            connected=connected,
            unconnected_send=unconnected_send,
            route_path=route_path,
            data_type=SHORT_STRING,
        )
        print(f'... Got: {resp.value if resp else "EMPTY"}')

for i in range(8):
    test(False, f'bp/{i}', True)
``` 

With connected, so route_path and unconnected_send are ignored:
```
Running with: connected=True, route_path='bp/0', unconnected_send=False...
... Got: 1756-ENBT/A
Running with: connected=True, route_path='bp/1', unconnected_send=False...
... Got: 1756-ENBT/A
Running with: connected=True, route_path='bp/2', unconnected_send=False...
... Got: 1756-ENBT/A
Running with: connected=True, route_path='bp/3', unconnected_send=False...
... Got: 1756-ENBT/A
Running with: connected=True, route_path='bp/4', unconnected_send=False...
... Got: 1756-ENBT/A
Running with: connected=True, route_path='bp/5', unconnected_send=False...
... Got: 1756-ENBT/A
Running with: connected=True, route_path='bp/6', unconnected_send=False...
... Got: 1756-ENBT/A
Running with: connected=True, route_path='bp/7', unconnected_send=False...
... Got: 1756-ENBT/A
```
--- and ---
```
Running with: connected=True, route_path='bp/0', unconnected_send=True...
... Got: 1756-ENBT/A
Running with: connected=True, route_path='bp/1', unconnected_send=True...
... Got: 1756-ENBT/A
Running with: connected=True, route_path='bp/2', unconnected_send=True...
... Got: 1756-ENBT/A
Running with: connected=True, route_path='bp/3', unconnected_send=True...
... Got: 1756-ENBT/A
Running with: connected=True, route_path='bp/4', unconnected_send=True...
... Got: 1756-ENBT/A
Running with: connected=True, route_path='bp/5', unconnected_send=True...
... Got: 1756-ENBT/A
Running with: connected=True, route_path='bp/6', unconnected_send=True...
... Got: 1756-ENBT/A
Running with: connected=True, route_path='bp/7', unconnected_send=True...
... Got: 1756-ENBT/A
```

Unconnected and unconnected send works as expected:
```
Running with: connected=False, route_path='bp/0', unconnected_send=True...
... Got: EMPTY
Running with: connected=False, route_path='bp/1', unconnected_send=True...
... Got: 1756-EN2TR/C
Running with: connected=False, route_path='bp/2', unconnected_send=True...
... Got: EMPTY
Running with: connected=False, route_path='bp/3', unconnected_send=True...
... Got: 1756-ENBT/A
Running with: connected=False, route_path='bp/4', unconnected_send=True...
... Got: 1756-ENBT/A
Running with: connected=False, route_path='bp/5', unconnected_send=True...
... Got: EMPTY
Running with: connected=False, route_path='bp/6', unconnected_send=True...
... Got: 1756-DNB/E
Running with: connected=False, route_path='bp/7', unconnected_send=True...
... Got: EMPTY
```

ottowayi, 2023-05-17:
Oh wait, I totally missed it working with connected but failing with unconnected.  I didn't realize some services are connection-type dependent, but was able to replicate it with some ENBTs I have.  Unfortunately, there isn't a way to send a connected message to a different route than the one setup by the driver using `generic_message()`.  You'll have to create a new driver for each:

```
for slot in range(<slot_count>):
    with CIPDriver(f'192.168.5.11/1/{i}') as driver:  
         result = driver.generic_message(...)
```

tlf30, 2023-05-17:
I too had no idea that there were connection dependent services until I started trying to get port class info.
I have been playing with different types of cards, right now the best example of a card that has strange behavior is the 1756-DNB, which will respond to a request on the port class from the devicenet port, but not from the backplane.

For example, if I have an EWEB as my access point to the rack on 192.168.5.11, and the 1756-DNB in slot 11:
It returns error 0x05 when using a forward open via path: `192.168.5.11/bp/11`
But works if I go back through the DeviceNet port: `192.168.5.11/bp/11/2/0` (The card has a DeviceNet address of 0).

I am thinking I am running into either firmware bugs or intentional restrictions on the cards rather than a CIP messaging issue. 

I played a bit with building path specific connected messages using the generic message function. It seems that some cards like the EWEB and L6x processors do work, while others still have strange issues. I also have come to the conclusion that the L5x processors just simply don't implement many of the CIP classes that I would expect them to, which is strange as they actually have a lot of CIP classes that I did NOT expect them to.
My current workaround it to try to hit the port service, and if I cannot then I brute force try to find different port specific classes. Currently I am testing for:
```
* 0x66 - Backplane info / rack size
* 0xF5 - TCP/PI (Also includes 0xF6 if 0xF5 is found)
* 0xAA - Old 1756-ENET ethernet info
* 0x6F - Serial port
* 0xA2 - DF1 Port (Only if 0x6F is found?)
* 0xF0 - ControlNet (Run Identity request from nodes 1 to 99)
* 0x03 - DeviceNet (Run Identity request from nodes 0 to 63)
* 0xA6 - DHRIO: 0x64 (ExtendedDevice - Number of slots), 0x66 (ICP - Rack size), 0xA5 (DH+), 0xA4 (Routing Table)
```

I am sure there is a better way, but I actually observed RSLinx taking the same approach with the older ControlLogix cards, hence how I got the idea and have been figuring out which classes to query. 
Thoughts?

ottowayi, 2023-05-17:
I think you're finding a lot of the same weird behavior I was with these objects/services/devices not too long ago.  Are you also building something for discovery/mapping?  It seemed like the older firmwares were much less _compliant_ than the newer stuff and just weird things like devicenet devices didn't support get_attributes_all for the identity object.  

This library originally only cared about reading/writing tags, so the generic CIP stuff is not as user friendly.  Like `generic_message` was something I used internally and decided to make it a public method later on, which is kind of why you really have to know what the args mean.  I've been working on a new version where I'm rewriting the entire protocol implementation from scratch.  I think it would make this type of stuff easier, but I haven't had much time to work on it recently so I have no timeline on when it will be done. 

One tip for DeviceNet, you can read attribute 13 to get the active node table.  This is a USINT[8] array where each bit corresponds to a node address, any True bits are active.  So you can narrow it down a bit and not have to scan every address.

tlf30, 2023-05-17:
Yes, that is exactly what I am doing.

> One tip for DeviceNet, you can read attribute 13 to get the active node table. This is a USINT[8] array where each bit corresponds to a node address, any True bits are active. So you can narrow it down a bit and not have to scan every address.

Very good to know, thank you!

tlf30, 2023-05-24:
I have everything mostly working. I see the eweb module has some issues that are specific to that module (I'm going to assume bugs in the firmware) 
Thank you for the help!

closed 2023-05-24 by tlf30