ottowayi/pycomm3 #299: [BUG] - Domain name not supported as a valid IPAddress

alliehunter, 2023-08-22:
**Description**
Pycomm3 does not support passing in a domain name as a valid IPAddress for CIPDriver. Specifically, using: `ipaddress.ip_address(ip)` when passing in a domain name results in a `ValueError`. 

We are using domain names to identify running docker containers that host an E/IP server that we want pycomm3 to communicate with, but pycomm3 is unable to identify the IP address since it is being passed in as a domain name instead of an `IPv4Address`.

**Code Sample**
One way to implement this is using the `socket` library: 
```python
ip_address = socket.gethostbyname(ip)
```

ottowayi, 2023-08-25:
You're right, this is a completely arbitrary and unnecessary limitation. If you would like to submit a pr to remove it, I'll gladly accept it. Else, I'll try to do it soon; hopefully this weekend, but can't make any promises.

ottowayi, 2023-08-27:
Added in v1.2.14 (#300)

closed 2023-08-27 by ottowayi

aregcoretigo, 2023-08-27:
Hi,

@alliehunter I am looking for a way to create an EIP server in a docker container like you described in this issue, could you share what frameworks\libs you are using to do that?
Thanks very much for your help.

Alex