ottowayi/pycomm3 #303: [BUG] -

AndrewStuhr, 2023-10-26:
While trying to read tags out of a 1756-L83E/B, the following error occurs:

Traceback (most recent call last):
  File "deviceTags.py", line 3, in <module>
    with LogixDriver('192.168.96.2/0') as plc:
  File "C:\Program Files\DAQuery\webserver\venv\lib\site-packages\pycomm3\clx.py", line 170, in __init__
    self._micro800 = self._list_identity().startswith(MICRO800_PREFIX)
AttributeError: 'NoneType' object has no attribute 'startswith'

Here is the code
`from pycomm3 import LogixDriver

with LogixDriver('192.168.96.2/0') as plc:
    print(plc.tags)
`

**The version of pycomm3 we're running is 0.8.1**, the reason being that newer versions return results that aren't JSON serializable.

Apologies if this issue has already been fixed, I didn't see any previous submissions regarding this issue.

As always, thanks for the wonderful library!

ottowayi, 2023-11-03:
hmm that's weird, it's hard to tell without more context/logs. The current code looks a bit different now, so it may have been fixed.  

Is it the `plc.tags` that isn't json-serializable? because there is the `plc.tags_json` property that should be.  I would recommend updating and trying that first.

AndrewStuhr, 2023-11-03:
I'll try again with the most recent pycomm3 versions and report back.

I'll also try out plc_.tags_json - thanks!

AndrewStuhr, 2023-11-17:
There was no issue reading the PLC using pycomm3==1.2.14, and the JSON tags worked great.  Thanks!

closed 2023-11-17 by AndrewStuhr