ottowayi/pycomm3 #317: Generic Messaging Help

mikeadkins, 2024-04-30:
I am attempting to use Pycomm3 with a Japanese PLC brand called Toyopuc. I have gotten the reading of data working properly with generic messaging but am having trouble with the sending of data. I keep getting an error in the log stating: Generic message 'generic' failed: Insufficient command data. Would anyone have any idea as to what is happening? Thanks for any help you can provide.

EDS file: `$ EZ-EDS Version 3.21.1.20161208 Generated Electronic Data Sheet

[File]
    $ File Description Test
        DescText = "PC10G-CPU EDS File";
    $ File Creation Date
        CreateDate = 05-18-2017;
    $ File Creation Time
        CreateTime = 16:22:28;
    $ Last Modification Date
        ModDate = 05-18-2017;
    $ Last Modification Time
        ModTime = 16:24:37;
    $ EDS Revision
        Revision = 1.1;

[Device]
    $ Vendor ID
        VendCode = 130;
    $ Vendor Name
        VendName = "JTEKT Corporation";
    $ Product Type
        ProdType = 12;
    $ Product Type String
        ProdTypeStr = "Communications Adapter";
    $ Product Code
        ProdCode = 1001;
    $ Major Revision
        MajRev = 1;
    $ Minor Revision
        MinRev = 1;
    $ Product Name
        ProdName = "PC10G-CPU";

[Device Classification]
        Class1 = EtherNetIP;

[Params]
        Param1 =
                0,                      $ reserved, shall equal 0
                ,,                      $ Link Path Size, Link Path
                0x0000,                 $ Descriptor
                0xC8,                   $ Data Type
                4,                      $ Data Size in bytes
                "RPI",                  $ name
                "microsecond",          $ units
                "",                     $ help string
                2000,10000000,50000,    $ min, max, default data values
                ,,,,                    $ mult, div, base, offset scaling
                ,,,,                    $ mult, div, base, offset links
                ;                       $ decimal places
        Param2 =
                0,                      $ reserved, shall equal 0
                ,,                      $ Link Path Size, Link Path
                0x0000,                 $ Descriptor
                0xC7,                   $ Data Type
                2,                      $ Data Size in bytes
                "Produced Data Size",   $ name
                "byte",                 $ units
                "",                     $ help string
                1,1448,2,               $ min, max, default data values
                ,,,,                    $ mult, div, base, offset scaling
                ,,,,                    $ mult, div, base, offset links
                ;                       $ decimal places
        Param3 =
                0,                      $ reserved, shall equal 0
                ,,                      $ Link Path Size, Link Path
                0x0000,                 $ Descriptor
[... 181 lines cut ...]
                "20 F5 24 01",
                2,
                0;

[Capacity]
        MaxCIPConnections = 128;
        MaxConsumersPerMcast = 128;
        TSpec1 = TxRx, 1, 1000;
        TSpec2 = TxRx, 1448, 1000;

`
My code: `drive_path = '192.168.250.25'
with CIPDriver(drive_path) as drive:    
    param=drive.generic_message(
    service=Services.set_attribute_single,
    class_code=ClassCode.assembly,
    instance=101,
    attribute=b'\x03',
    request_data= b'\x0f',
    data_type=BYTE,
    name='generic',
    connected=True,
    unconnected_send=True,
    route_path=True
    )`

mikeadkins, 2024-05-01:
Toyopuc does not use Tags. All of the data is mapped to what they call 'Link Bits'. L000-L01F. I have 2 bytes for assembly 100 and 2 bytes for assembly 101. I have used another device (Omron ERT1 remote I/O) and it works. I receive an INT that has the data for the 16 bits of each assembly.

mikeadkins, 2024-05-01:
I wasn't but I am now. I believe it is in the PLC side. If I monitor with Wireshark it does show success with set_attribute_single and it shows the data I sent (0x0a for this example).
[image]

I did find in the Toyopuc manual the following comment regarding Transport  Class. It supports Class 1 and Class 3 is not supported. I do not know enough about this subject to say if that is the problem. 

Also if I do a get_attribute_single on the same assembly 101 after the set I can see the data I sent, but it never shows up in the PLC monitor. This brand of PLC is very outdated and only used in one auto manufacturer who also owns the PLC company so it wouldn't surprise me if it doesn't work. Thanks.

backrent, 2024-09-26:
If it's not showing on the monitor, you could try linking the bit to a counter.  Maybe it only goes high for 1 scan.